SofaDefrost/SoftRobots #7: Visualization bug with the PneunetGripper-Tutorial example

quentinfrancois0, 2018-03-26:
Hello everyone,

I just ran the PneunetGripper-Tutorial and specifically the step 7 of it and I get a visualization bug. Indeed I only have the left-bottom corner of the window which is active. You can see that the plane and the fingers are cut on the screenshot.

Here is the screen : https://imgur.com/a/Kt1Ts

I'm on MacOS 10.13.3 with Sofa 17.12

EulalieCoevoet, 2018-03-27:
Hi Quentin, 
I don't have this issue on my computer... Maybe someone using MacOS can help @bcarrez...

EulalieCoevoet, 2018-09-10:
Hello Quentin,
Sorry no one helped you...
Do you still have the problem?  
Or should we close the issue?

quentinfrancois0, 2018-09-10:
Hello Eulalie,

I juste launched the scene to see if there still is the issue and yes nothing has changed in the behavior of the screen. I found that it depends on the screen resolution. The bug occurs on a 4K screen but when I drag the sofa window on a 1080p screen I don't have the issue. Perhaps it will help.

EulalieCoevoet, 2018-09-27:
I think it would be great to open an issue about that in Sofa. 
I will close this one because I don't think it is related to the plugin.

closed 2018-09-27 by EulalieCoevoet

damienmarchal, 2018-09-27:
I sound also very plausible to me this is related to runSofa more than softrobots. But I have no 4k CG/monitor.