SofaDefrost/SoftRobots #42: Errors in Running PneunetGripper Example

asalimil, 2019-06-10:
Hi,

I have an issue when I want to run the python codes for PneunetGripper example in SOFA scene. I just added "SoftRobots"  and "STLIB" plugins to SOFA and SOFA_Pyhton plugin  as well. 

I followed the tutorial in which coding for PneunetGripper example is explained step by step (step 1 to 7) and accumulate all the python codes for each step in a single *.pyscn. Then, I opened the resulted *.pyscn from SOFA, but I cannot see anything except the "coordinate system" and a "red plate" in the screen. 

Moreover, I have tried each step separately to open in SOFA. I could be able to see the mesh for Step 1, ForceField and Mass in Step 2, the stiff layer in Step 3, the boundary condition (constraint) in Step 4, but I would not see anything when I run the whole python code which is also mentioned in the Appendix of the tutorial for Pneunetgripper. 

I got these errors:
1. Plugin not found: "SoftRobots"
2. RuntimeError: Unable to create 'directSolver' of type 'SparseLDLSolver' in node 'finger'. Object type SparseLDLSolver<> was not created. This component has been PLUGINIZED since SOFA v18.06. To continue using this component you need to update you scene and add <RequiredPlugin pluginName='SofaSparseSolver'/>

The directory in which I have the pythons codes and sofa:
/home/sofa/src&build/
/home/Python_Sofa/PneunetGripper/details/pneunetgripper-tuto.pyscn

I will be appreciated if you help me to solve these issues.
Thank you,
Amir

damienmarchal, 2019-06-11:
Hi amir, 

Glad to hear about your interest in SoftRobots

Can you tell me which version of Sofa are you using ? Did you compile it yourself ? Use the baseline binary from sofa-framework or our custom binary Defrost is providing ?

asalimil, 2019-06-11:
Hi @damienmarchal 
Thank you for your response.
I have installed the unstable version of SOFA (the latest version) and followed the installation procedure mentioned in these two links:
https://project.inria.fr/softrobot/install-get-started-2/download/
https://project.inria.fr/softrobot/install-get-started-2/building-on-linux/
I did these steps:
1. Install the prerequisites for SOFA according to the link:
https://www.sofa-framework.org/community/doc/getting-started/build/linux/

2. I have made the required directories 
sofa/
      src/
      build/

3. >> git clone -b master https://github.com/sofa-framework/sofa.git sofa/master/src/

4. Used Cmake to build SOFA
I have added /home/username/Qt/5.12.2/gcc_64 to the path
I did not use ninja, so I used Codeblocks - Unix CMake
Configure the project 2 times and then Generate

4. In the /build directory I used >> make in order to make the project

5. >> git clone https://github.com/SofaDefrost/SoftRobots

6. Open Cmake again and add SOFA_External_Directories and add the path in which my SoftRobots folder is located: /home/username/SoftRobots/
Check the PLUGIN_SOFAPYTHON
Configure two times, Generate again

8. In the build directory >> make (again)

9. Open SOFA in /bin directory by: ./runSofa

10. Checked if the  SofaPython plugin is added to SOFA.
SOFA > Edit > Plugin Manager > Add > libSofaPython.so

11. Opened the example from the SoftRobots plugin folder: /home/username/SoftRobots/docs/tutorials/PneunetGripper/details/step7-grabTheCube.pyscn

But, I cannot see anything except the coordinate frame and the red plane. I got the errors with Finger 1, Finger 2, and Finger 3.
[image]

I will be appreciated if you let me know how to solve this issue.

Thank you,
Amir

EulalieCoevoet, 2020-10-22:
I've run this tutorial and it's working fine. I guess the problem has been solved? 
I'll close the issue. You can reopen it if for some reason it's not working on your computer.

closed 2020-10-22 by EulalieCoevoet