SofaDefrost/SoftRobots #48: CableGripper tutorial not working

mthieffry, 2020-02-21:
CableGripper tutorial is not working with current version of Sofa (only step1 is working).
Here is the error obtained when launching step0 : 

[ERROR]   [PythonScript] TypeError: 'staticmethod' object is not callable
  File "step0.pyscn", line 11, in createScene
    Gripper(rootNode)
  File "gripper.py", line 14, in Gripper
    pullPointLocation=[3, 10.5, 3])
  File "finger.py", line 39, in Finger
    translation=translation)

STLIB and SoftRobots plugins are up to date, I tested with Sofa-defrost and Sofa-framework sources.

I imagine that the problem is not very complicated but I let the professionals take care of that.

VannesteFelix, 2020-02-21:
Its probably due to this : https://github.com/SofaDefrost/STLIB/commit/572d068020e124f638010e24bdabaaeb3db4df17.
Can you merge it in STLIB master if its correct for you @damienmarchal ?

damienmarchal, 2020-04-29:
You are probably right felix..merged.

closed 2020-04-29 by damienmarchal