SofaDefrost/SoftRobots #49: Unable to create 'HexahedronFEMForceField'

mthieffry, 2020-03-18:
Hi,
Are there SOFA experts here who are forced to stay at home and do not know what to do?

I was trying to create a simple to test some algorithms. I would like to create a model a beam using hexahedra model.
However, I am not able to use the "HexahedronFEMForceField" component properly.
Here is the error I get:

```
beam.createObject('HexahedronFEMForceField', youngModulus=yModulus, poissonRatio = pRatio, method="large")    
```

gives the following error:

```RuntimeError: Unable to create 'HexahedronFEMForceField' of type 'HexahedronFEMForceField' in node 'beam'. Object type HexahedronFEMForceField<> was not created  
 The object is in the factory but cannot be created.  
Requested template: ()  
Available templates: Vec3d,
```

when I add the template in the component call I get a slightly different error:

```
beam.createObject('HexahedronFEMForceField', template='Vec3d',youngModulus=yModulus, poissonRatio=pRatio, method="large")    

RuntimeError: Unable to create 'HexahedronFEMForceField' of type 'HexahedronFEMForceField' in node 'beam'. Object type HexahedronFEMForceField<Vec3d> was not created  
 The object is in the factory but cannot be created.  
Requested template: Vec3d(Vec3d)  
Available templates: Vec3d,  
```

What did I do wrong ?!

Thanks !
Maxime

------

Here is the complete scene : 

```
from stlib.scene import MainHeader

totalMass=0.01
pRatio=0.3
yModulus=500

def createScene(rootNode):
                MainHeader(rootNode, gravity=[0.0,-981.0,0.0])
                rootNode.createObject('RequiredPlugin', pluginName='SoftRobots')
                rootNode.createObject('RequiredPlugin', pluginName='SofaPython')
                rootNode.createObject('RequiredPlugin', pluginName='SofaPreconditioner')

                beam = rootNode.createChild('beam')
                beam.createObject('EulerImplicitSolver', name="odesolver", rayleighStiffness=0.1, rayleighMass=0.1)
                beam.createObject('ShewchukPCGLinearSolver', name='linearSolver',iterations=200, tolerance=1.0e-18, preconditioners="preconditioner")
                beam.createObject('RegularGridTopology', name="SoftBeam", nx="6", ny="2", nz="2", min="0 0 0", max="1 0.2 0.2")     
      
                beam.createObject('HexahedronSetGeometryAlgorithms')
                beam.createObject('HexahedronFEMForceField', template='Vec3d',youngModulus=yModulus, poissonRatio=pRatio, method="large")    
                beam.createObject('MechanicalObject', name="meca")
                beam.createObject('DiagonalMass', totalMass=totalMass)

                beam.createObject('SparseLDLSolver', name="preconditioner")
                beam.createObject('LinearSolverConstraintCorrection', solverName="preconditioner")      

                beam.createObject('BoxROI', name="boxROI", box="-0.01 -0.01 -0.01 0.1 0.2 0.2", drawBoxes="true")
                beam.createObject('RestShapeSpringsForceField', points='@boxROI.indices', stiffness='1e12')

                beam.createObject('GenericConstraintCorrection', name='GCS', solverName='preconditioner')

                return rootNode      
```

EulalieCoevoet, 2020-03-18:
The fix is to put the `FEMForceField` after the `MechanicalObject` (and note that you have two `ConstraintCorrection`).

olivier-goury, 2020-03-19:
I agree with Eulalie! And also, maybe you don't need `HexahedronSetGeometryAlgorithms`.
There is a similar example in ModelOrderReduction/tools/sofa_test_scene/hexaBeam.pyscn .
Happy confinement!

mthieffry, 2020-03-19:
Eulalie was right (as always...)
Thanks !

closed 2020-03-19 by mthieffry