SofaDefrost/SoftRobots #50: Build Error

CLiu-eng, 2020-04-25:
I tried to use source code to install SOFA and SoftRobots plugin on Mac OS 10.15.4.
I got the unstable version of SOFA by `git clone -b master https://github.com/sofa-framework/sofa.git sofa/master/src/`

Then I download the Plugin SoftRobots 17.12 version (I also got one version from `git clone https://github.com/SofaDefrost/SoftRobots`, but when I configured it in CMake GUI, it always reported error). The 17.12 version of SoftRobot configured well then I generated the whole file.

However, when I tried to build it and after a long wait, it reports errors as below. The screenshot is not complete but other errors are similar to these.
[image]

It seems that there's something wrong with the softrobots plugin in the source code. I am not sure what caused this problem. Is it because the versions of SOFA and SoftRobots Plugin are not compatible? Has anyone met this before? How can I solve this problem?

EulalieCoevoet, 2020-10-20:
Hi CLiu-eng,

Sorry no one helped you yet. Did you manage to fix the problem? 
I think for building problems it is better to ask on Sofa forum. There are already SoftRobots related topics there.

CLiu-eng, 2020-10-21:
Thank you for following up with my question and it has been sorted.

EulalieCoevoet, 2020-10-22:
Okay great.

closed 2020-10-22 by EulalieCoevoet