SofaDefrost/SoftRobots #54: First steps tutorial for sofa and softrobotics

MADRobot69, 2020-05-24:
Hi, I've been trying to run the first steps tutorial, and after of days of trying to install sofa/other installations required in the right places, I finally managed to open the first steps tutorial yesterday. However, after editing, I saved and reloaded the page, but nothing happened. I shut it down and for some reason everything glitched out. I've reinstalled sofa and the soft robotics plugin, and now when I right lick on DefaultVisualManagerLoop, a pop up appears for a split second then disappears. I've reinstalled python 2.7, I'm just completely lost on what to do next/ what's going wrong, is this something anyone has experienced before? Any help would be greatly appreciated, thanks!

MADRobot69, 2020-05-24:
Update: I managed to do the first part of the tutorial, but then I started experiencing the same problem - the software crashed so I had to restart runSofa, then it wouldn't allow me to 'write it myself', I can just select the options to 'try the scene in Sofa.'

damienmarchal, 2022-05-12:
Thank for reporting @MADRobot69,

Given that new version are now available I close the issue. Please report if this still happens with new version.

closed 2022-05-12 by damienmarchal