SofaDefrost/SoftRobots #71: Problem with PointCloudProcessing

pasqualeferr94, 2021-03-18:
Hello I recently want to use Soft Robots with camera plugin ( I have a Realsense D435) and I found an error in the code PointCloudProcessing.cpp, The following:
`home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudProcessing.cpp: In member function ‘void sofa::component::pointcloudprocessing::PointCloudProcessing::computeEffectorsDirection(pcl::PointCloud<pcl::PointXYZRGB>::Ptr&, pcl::PointCloud<pcl::Normal>::Ptr&)’:
/home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudProcessing.cpp:272:53: error: declaration of ‘pcl::PointCloud<pcl::Normal>::Ptr cloud_normals’ shadows a parameter
  272 |     pcl::PointCloud<pcl::Normal>::Ptr cloud_normals (new pcl::PointCloud<pcl::Normal>);
      |                                                     ^
/home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudProcessing.cpp:246:146: warning: unused parameter ‘cloud_normals’ [-Wunused-parameter]
  246 | YZRGB>::Ptr& robot_effectors, pcl::PointCloud<pcl::Normal>::Ptr& cloud_normals)
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~

In file included from /usr/include/eigen3/Eigen/Core:366,
                 from /usr/include/pcl-1.10/pcl/pcl_macros.h:75,
                 from /usr/include/pcl-1.10/pcl/point_types.h:42,
                 from /home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudStreaming.h:34,
                 from /home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudProcessing.h:28,
                 from /home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudProcessing.cpp:25:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<double, 4, 4>; Src = Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<float>, Eigen::Matrix<float, 4, 4> >; Func = Eigen::internal::assign_op<double, float>]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:804:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<double, 4, 4>; Src = Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<float>, Eigen::Matrix<float, 4, 4> >; Func = Eigen::internal::assign_op<double, float>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:782:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<double, 4, 4>; Src = Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<float>, Eigen::Matrix<float, 4, 4> >]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:714:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<float>, Eigen::Matrix<float, 4, 4> >; Derived = Eigen::Matrix<double, 4, 4>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_identity_op<float>, Eigen::Matrix<float, 4, 4> >; _Scalar = double; int _Rows = 4; int _Cols = 4; int _Options = 0; int _MaxRows = 4; int _MaxCols = 4]’
/home/pasquale/sofa/src/applications/plugins/SoftRobots/src/SoftRobots/component/camera/PointCloudProcessing.cpp:121:45:   required from here
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:834:3: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
  834 |   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [applications/plugins/SoftRobots/CMakeFiles/SoftRobots.dir/build.make:362: applications/plugins/SoftRobots/CMakeFiles/SoftRobots.dir/src/SoftRobots/component/camera/PointCloudProcessing.cpp.o] Errore 1
make[1]: *** [CMakeFiles/Makefile2:8932: applications/plugins/SoftRobots/CMakeFiles/SoftRobots.dir/all] Errore 2
make: *** [Makefile:163: all] Errore 2

 `

How to resolve this issue?
Anyway, Want basically track my robot with a camera.... I am also buying the license of SoftRobot.Inverse plugin, the tracking is possible in the Inverse plugin or in this one?

Thanks in Advance

pasqualeferr94, 2021-04-20:
Hello @zhongkaidefrost  Can you help me in this issue?
Thanks a lot!

EulalieCoevoet, 2021-05-14:
Hi @pasqualeferr94,

Sorry for this late reply. @zhongkaidefrost made a [PR](https://github.com/SofaDefrost/SoftRobots/pull/75) regarding your issue. Could you try his changes and tell us if it works for you?

Thanks, 
Eulalie

pasqualeferr94, 2021-05-14:
Hi @EulalieCoevoet ,
Thanks a lot, the compilation issue is resolved!
Now I have just some question on how to use this  part of the plugin.
Now I explain what i want to do.
I have a realsense D435 RGBD camera. I want to connect this working in parallel with sofa. In particular i wan that if have an object, let's say a cube with a marker (or Aruco marker or just a colored one) , can i make appear the mesh (a triangular one), of the object in the scene only when the marker is detected by camera. If the camera tracks the position of the marker, the sofa scene can also repeat the displacement?

I hope that the question is clear, thanks already for the help!

EulalieCoevoet, 2021-05-14:
Great! thanks.

As for your question, I'm sorry I don't really know this part of the plugin... I'll ask Zhongkai. I'll keep you updated!
Since the issue has been resolved, I'll close it.

closed 2021-05-14 by EulalieCoevoet