SofaDefrost/SoftRobots #76: SOFA_v20.12.00_Linux tutorial plugins

akruszew, 2021-06-03:
The softrobot &co plugins to have to be added one by one after a fresh install (SOFA_v20.12.00_Linux). It could be nice to update the default configuration in the binaries.

EulalieCoevoet, 2021-11-02:
A good practice is to use the component `RequiredPlugin` in your scene. It has two advantages:
1- it will load the plugin
2- it tells to potential other users that your scene needs the plugin 

I believe most scenes in SoftRobots/docs have it, so there is no need to load the plugins using runSofa/Edit/PluginManager if that's what you mean.

EulalieCoevoet, 2022-03-17:
However you cannot use the RequiredPlugin approach for SofaPython3, since the plugin is needed to load the python file. 
This has been updated in the new release SOFA_v21.12.00_* that will be available in the coming days.

closed 2022-03-17 by EulalieCoevoet