SofaDefrost/SoftRobots #86: Dependence of SurfacePressureConstraint on dt with pressure control (valueType="presure")

cbschaff, 2021-09-16:
Hey. I noticed a strange dependence between the SurfacePressureConstraint and the simulator time step. I noticed this in one of my scenes and found that it affects the examples in this repo too.

The scene I used is a modified version of [this example](https://github.com/SofaDefrost/SoftRobots/blob/master/docs/examples/component/constraint/SurfacePressureConstraint/SurfacePressureConstraint.pyscn) provided by the SoftRobots plugin.

The script has been edited to increase pressure from 0 to 4 (kpa?) over a ramp up period of 0.2 seconds.

The provided videos show the behavior of the stanford bunny under 3 different dt's: 0.01, 0.001, 0.0001.
I provided an additional video with dt=0.0001 which increases pressure more slowly (over a period of 1 second).

A zip file containing the modified script and the videos can be found here:
[bunny_dependence_on_dt.zip](https://github.com/SofaDefrost/SoftRobots/files/7180216/bunny_dependence_on_dt.zip)

The videos show:
- The bunny deforms differently at each dt.
- With dt=0.0001, the bunny explodes around 0.5 kpa, independent of how fast the pressure is increased.

Note that the speed/length of the videos don't correspond with real time. 

https://user-images.githubusercontent.com/7145111/133666563-b85c0dab-adb1-419f-a040-3a6d6c38af4d.mp4

https://user-images.githubusercontent.com/7145111/133666561-098e2ec2-82a5-4799-a4fc-b53dfd563253.mp4

https://user-images.githubusercontent.com/7145111/133666559-7e6f004d-c6d0-40e3-b46e-7211511bf74b.mp4

https://user-images.githubusercontent.com/7145111/133666556-9b9a74ea-d081-4d6e-ad4b-7daf042f3cbc.mp4

EulalieCoevoet, 2021-09-17:
Hi @cbschaff, 

You are right, there is a time step dependency. You should divide the value by dt to give/get the real pressure. 
The same applies to the other constraint components. 

It's something that we've been aware of for a while now... I'm gonna reopen the discussion and hopefully we will choose between adding the info or changing the beviahor.

hugtalbot, 2021-09-17:
thanks @EulalieCoevoet for the feedback!
hope it helps @cbschaff

cbschaff, 2021-09-17:
Thanks for the quick response!

damienmarchal, 2022-05-12:
@EulalieCoevoet Can we close the issue ?

EulalieCoevoet, 2022-05-12:
@damienmarchal well it's still an issue, not solved yet. I'd rather keep it open.

EulalieCoevoet, 2022-06-16:
Warnings have been added. I'll close this issue.

closed 2022-06-16 by EulalieCoevoet

iman-adibnazari, 2024-06-12:
Hello,

I wanted to follow up on this issue and see if the timestep dependence is still present and, if so, if it is linear (i.e., if I increase my timestep by a factor of 10, then scaling pressure inputs by a factor of 10 should result in the same net behavior). I'm running into a similar issue but I'm unsure if its due to simulation stability being affected by the size of my timestep or if it's due to the need to consider the timestep dependence of this surface pressure constraint.