SofaDefrost/SoftRobots #142: Error prevent to fully load firststeps-tuto.py

alxbilger, 2022-05-09:
File to open: docs/sofapython3/tutorials/FirstSteps/firststeps-tuto.py

The error is:
```
[ERROR]   [OglModel(unnamed)] Source object "@loader" NOT FOUND.
[ERROR]   [SofaPython3::SceneLoader] Unable to completely load the scene from file 'docs\sofapython3\tutorials\FirstSteps\firststeps-tuto.py'.  
Python exception:   
  ValueError: 4:Source object "@loader" NOT FOUND.
```

It seems that the OglModel defined in visualmodel.py from STLIB cannot be created from a loader.

EulalieCoevoet, 2022-05-09:
I think we should move this discussion to [SofaPython3](https://github.com/sofa-framework/SofaPython3), it seems to be an issue with Sofa.Prefab. Commenting one of those lines for instance fixes the issue:

```diff
--- a/Plugin/src/SofaPython3/Prefab.cpp
+++ b/Plugin/src/SofaPython3/Prefab.cpp
@@ -81,7 +81,7 @@ void Prefab::doReInit()
 Prefab::Prefab()
 {
     m_filelistener.m_prefab = this;
-    m_datacallback.addCallback( std::bind(&Prefab::reinit, this) );
+//    m_datacallback.addCallback( std::bind(&Prefab::reinit, this) );
 }
 
 
@@ -97,7 +97,7 @@ void Prefab::addPrefabParameter(const std::string& name, const std::string& help
     {
         sofa::core::objectmodel::BaseData* data = sofapython3::addData(py::cast(this), name, py::none(), defaultValue, help, "Prefab's properties", type);
         data->setRequired(true);
-        m_datacallback.addInputs({data});
+//        m_datacallback.addInputs({data});
     }
 }
```

EulalieCoevoet, 2022-05-17:
After discussing with Damien, the real issue comes from how the properties are used in the prefab `VisualModel` from STLIB. So I'll close this issue to open one in STLIB.

closed 2022-05-17 by EulalieCoevoet