SofaDefrost/SoftRobots #144: SoftRobots plugin must not refer SoftRobots.Inverse plugin

alxbilger, 2022-05-17:
A lot of scenes from the SoftRobots plugin refer the SoftRobots.Inverse plugin through the `RequiredPlugin` component. Yet, the SoftRobots.Inverse plugin depends on SoftRobots, but not the opposite.

- If the scene does not need SoftRobots.Inverse, it should be removed from the required plugin lists
- If the scene does need SoftRobots.Inverse, the scene should be: 1) transferred to the SoftRobots.Inverse plugin, or 2) transformed to remove the dependency.

EulalieCoevoet, 2022-06-16:
I went through the docs folder and didn't find any unnecessary `RequiredPlugin`. 
I still think that the inverse steps of our tutorials belong in this repository. However, I came across three other simulations using the inverse resolution. I've set it as an option (false by default). I think we should keep the option because in each case it's a really nice demonstration/feature/example of use.

EulalieCoevoet, 2023-10-20:
@alxbilger Should we close this issue ?

closed 2024-07-26 by EulalieCoevoet