SofaDefrost/SoftRobots #148: Connect with Solidworks model

ozilxu, 2022-06-03:
Hi， @alxbilger @guparan may I ask how I can import my Solidworks file into the sofa?

alxbilger, 2022-06-04:
@ozilxu I don't know about the file formats that Solidworks can export. Most of the mesh loaders are in this folder: https://github.com/sofa-framework/sofa/tree/master/Sofa/Component/IO/Mesh/src/sofa/component/io/mesh. You can check if one of the loaders is compatible with your file.

ozilxu, 2022-06-09:
> @ozilxu I don't know about the file formats that Solidworks can export. Most of the mesh loaders are in this folder: https://github.com/sofa-framework/sofa/tree/master/Sofa/Component/IO/Mesh/src/sofa/component/io/mesh. You can check if one of the loaders is compatible with your file.

Thanks for the help! I will try with this mesh loader!

closed 2022-06-16 by EulalieCoevoet