SofaDefrost/SoftRobots #150: Some issues reported in the tutorial of sofapython3

ozilxu, 2022-06-09:
@guparan @ErwanDouaille @alxbilger @hugtalbot @damienmarchal 

I have encountered several problems in the tutorial of sofapython3
## 1. When I try to change the color with the code in the **FirstStep** tutorial

```
Cube(rootNode,
      translation=[0.0,0.0,0.0],
      uniformScale=20.0,
      color=[0.0,0.0,1.0]) 
```

After reload, the color of the cube has not been changed and in the terminal shows **Finished validating node "root".
[WARNING] [CGLinearSolver(CGLinearSolver)] Required data "iterations" has not been set. Falling back to default value: 25
[WARNING] [CGLinearSolver(CGLinearSolver)] Required data "tolerance" has not been set. Falling back to default value: 1e-05
[WARNING] [CGLinearSolver(CGLinearSolver)] Required data "threshold" has not been set. Falling back to default value: 1e-05**

I am not sure if it is the problem of the code itself.

## 2.  When I try to use the keyboard to control the cable in the tutorial of **CableGripper**

The terminal shows **Finished validating node "root".
[WARNING] [Node(root)] Default Animation Manager Loop will be used. Add DefaultAnimationLoop to the root node of scene file to remove this warning
[INFO]    [SofaViewer] QtViewer::keyPressEvent, CONTROL pressed
[INFO]    [SofaViewer] QtViewer::keyPressEvent, CONTROL pressed**

But the cable is not moving, and I don't know the reason.

## 3. In the Pneunet Gripper tutorial

When I press ctrl++ or ctrl -, the gripper is not inflated or deflated, the terminal just shows **WARNING] [LocalMinDistance(Proximity)] Determinant is null** 

## 4. The detailed tutorials of CircularRobot, DiamondRobot，Trunk and SoftArmGripper are lost.

EulalieCoevoet, 2022-06-09:
1. `Cube` is expecting a list of four elements [R,G,B,A] for the color. I think if you give only [R,G,B], a 0 will be added at the end which will give you a transparent object (see this [issue](https://github.com/sofa-framework/SofaPython3/issues/254)). Could that be your problem? 
2. If the problem occurs in the step3 of the tutorial, it has been fixed on the master branch of SoftRobots. You can see [this discussion](https://github.com/SofaDefrost/SoftRobots/discussions/134). 
3. Sorry I can't reproduce the problem. In `PneunetGripper/details/fingerController.py` you can try to display the key and check for `+` and `-`:
```python
 def onKeypressedEvent(self, e):
        print(e['key'])
```
4. There are no detailed steps for these tutorials yet. We'll work on that :)
5. Which step is that?

ozilxu, 2022-06-13:
New problem when opening the tripod tutorial, the terminal shows:

**Global Bounding Box seems very small; Your viewer settings (based on the bbox) are likely invalid, switching to default value of [-1,-1,-1,1,1,1].This is caused by using component which does not implement properly the computeBBox function.You can remove this warning by manually forcing a value in the parameter bbox="minX minY minZ maxX maxY maxZ" in your root node 

########## SIG 11 - SIGSEGV: segfault ##########
  sofa::helper::BackTrace::sig(int)
  sofa::gui::qt::viewer::GLBackend::drawBackgroundImage(int, int)
  sofa::gui::qt::viewer::qgl::QtGLViewer::DisplayOBJs()
  sofa::gui::qt::viewer::qgl::QtGLViewer::drawScene()
  sofa::gui::qt::viewer::qgl::QtGLViewer::draw()
  QGLViewer::paintGL()
  QWidget::event(QEvent*)
  QApplicationPrivate::notify_helper(QObject*, QEvent*)
  QApplication::notify(QObject*, QEvent*)
  QCoreApplication::notifyInternal2(QObject*, QEvent*)
  QWidgetPrivate::sendPaintEvent(QRegion const&)
  QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::syncBackingStore()
  QWidget::event(QEvent*)
  QMainWindow::event(QEvent*)
  QApplicationPrivate::notify_helper(QObject*, QEvent*)
  QApplication::notify(QObject*, QEvent*)
  QCoreApplication::notifyInternal2(QObject*, QEvent*)
  QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
  g_main_context_dispatch
  g_main_context_iteration
  QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
  QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
  QCoreApplication::exec()
  sofa::gui::qt::RealGUI::mainLoop()
  sofa::gui::GUIManager::MainLoop(boost::intrusive_ptr<sofa::simulation::Node>, char const*)
  __libc_start_main
Segmentation fault (core dumped)
**

EulalieCoevoet, 2022-06-16:
Hi @ozilxu,

Could you give feedback on your first issues ?
As for the new issue, which scene gives you that ?

ozilxu, 2022-07-11:
@EulalieCoevoet @damienmarchal Thanks a lot for your help and sorry for my late reply! Now I am using the new version of 22.06 in my **Windows OS** embedded with **Python3.8** for testing.

1. My first issue of the color, when I try to use the format of the color **[R,G,B,A]**

`cube = Cube(rootNode, translation=[0.0,0.0,0.0],
                    uniformScale=20.0,color=[1.1,0.7,0.6,0.1])`
                    
The color cannot be changed as well and I got the below information.
```
[INFO]    [SceneCheckerVisitor] Finished validating node "root".
[WARNING] [CGLinearSolver(CGLinearSolver)] Required data "iterations" has not been set. Falling back to default value: 25
[WARNING] [CGLinearSolver(CGLinearSolver)] Required data "tolerance" has not been set. Falling back to default value: 1e-05
[WARNING] [CGLinearSolver(CGLinearSolver)] Required data "threshold" has not been set. Falling back to default value: 1e-05
```

2. When I try to add the below code in the DefrostSofaBundle_win64_python3.8_v22.06.00-beta4\plugins\SoftRobots\docs\sofapython3\tutorials\CableGripper\details\step3.py 

`rootnode.createObject('FreeMotionAnimationLoop')
rootnode.createObject('GenericConstraintSolver', tolerance=-1e-5, maxIterations=100)`

I cannot control the cable as well and the terminal shows the below information:
 
``` This scene is using component defined in plugins but is not importing the required plugins.
  Your scene may not work on a sofa environment with different pre-loaded plugins.
  To fix your scene and remove this warning you just need to cut & paste the following lines at the beginning of your scene (if it is a .scn):
  <RequiredPlugin name="Sofa.Component.AnimationLoop"/> <!-- Needed to use components [FreeMotionAnimationLoop] -->
  <RequiredPlugin name="Sofa.Component.Constraint.Lagrangian.Correction"/> <!-- Needed to use components [LinearSolverConstraintCorrection] -->
  <RequiredPlugin name="Sofa.Component.Constraint.Lagrangian.Solver"/> <!-- Needed to use components [GenericConstraintSolver] -->
  <RequiredPlugin name="Sofa.Component.Engine.Select"/> <!-- Needed to use components [BoxROI] -->
  <RequiredPlugin name="Sofa.Component.IO.Mesh"/> <!-- Needed to use components [MeshVTKLoader] -->
  <RequiredPlugin name="Sofa.Component.LinearSolver.Direct"/> <!-- Needed to use components [SparseLDLSolver] -->
  <RequiredPlugin name="Sofa.Component.Mapping.Linear"/> <!-- Needed to use components [BarycentricMapping] -->
  <RequiredPlugin name="Sofa.Component.Mass"/> <!-- Needed to use components [UniformMass] -->
  <RequiredPlugin name="Sofa.Component.ODESolver.Backward"/> <!-- Needed to use components [EulerImplicitSolver] -->
  <RequiredPlugin name="Sofa.Component.SolidMechanics.FEM.Elastic"/> <!-- Needed to use components [TetrahedronFEMForceField] -->
  <RequiredPlugin name="Sofa.Component.SolidMechanics.Spring"/> <!-- Needed to use components [RestShapeSpringsForceField] -->
  <RequiredPlugin name="Sofa.Component.StateContainer"/> <!-- Needed to use components [MechanicalObject] -->
  <RequiredPlugin name="Sofa.Component.Visual"/> <!-- Needed to use components [VisualStyle] -->
  <RequiredPlugin name="Sofa.GL.Component.Rendering3D"/> <!-- Needed to use components [OglSceneFrame] -->
  ```
  3. When I try with **DefrostSofaBundle_win64_python3.8_v22.06.00-beta4\plugins\SoftRobots\docs\sofapython3\tutorials\PneunetGripper\details\step7-withSTLIB.py** and try to control the arm with keyboard. I cannot see the deformation of the arm the the terminal shows:
```
  [WARNING] [SceneCheckUsingAlias] This scene is using hard coded aliases. Aliases can be very confusing, use with caution.
  - MeshOBJLoader: 3 created with alias "MeshObjLoader"
[INFO]    [SceneCheckerVisitor] Finished validating node "root".
[WARNING] [Node(root)] Default Visual Manager Loop will be used. Add DefaultVisualManagerLoop to the root node of scene file to remove this warning
[INFO]    [SofaViewer] QtViewer::keyPressEvent, CONTROL pressed
[INFO]    [SofaViewer] QtViewer::keyPressEvent, CONTROL pressed
 ```
  4. For the other part of the tutorial, like CircularRobot, DiamondRobot，Trunk and SoftArmGripper，are there website or tutorial for explaining the corresponding Python APIs?  
  
  5. **When I open the scene of duplicating the finger in the **CableGripper tutorial**, the scene is so weird to have two fingers unregularly crosses.** -> This issue is fixed with the new version and the duplication is worked! Thanks again!
  
  6. I still have this issue with the 22.06 linux version with Python3.8 when trying to opening any .py file with Sofa
   ```
  ########## SIG 11 - SIGSEGV: segfault ##########
  sofa::helper::BackTrace::sig(int)
  sofa::gui::qt::viewer::GLBackend::drawBackgroundImage(int, int)
  sofa::gui::qt::viewer::qgl::QtGLViewer::DisplayOBJs()
  sofa::gui::qt::viewer::qgl::QtGLViewer::drawScene()
  sofa::gui::qt::viewer::qgl::QtGLViewer::draw()
  QGLViewer::paintGL()
  QWidget::event(QEvent*)
  QApplicationPrivate::notify_helper(QObject*, QEvent*)
  QApplication::notify(QObject*, QEvent*)
  QCoreApplication::notifyInternal2(QObject*, QEvent*)
  QWidgetPrivate::sendPaintEvent(QRegion const&)
  QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
  QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
[... cut ...]
  QCoreApplication::notifyInternal2(QObject*, QEvent*)
  QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
  g_main_context_dispatch
  g_main_context_iteration
  QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
  QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
  QCoreApplication::exec()
  sofa::gui::qt::RealGUI::mainLoop()
  sofa::gui::GUIManager::MainLoop(boost::intrusive_ptr<sofa::simulation::Node>, char const*)
  __libc_start_main
Segmentation fault (core dumped)
 ```

EulalieCoevoet, 2023-10-20:
Hello @ozilxu, 

I'm really sorry we didn't follow up on these issues. I hope you managed to find solutions, hopefully the latest binaries worked for you ?

closed 2024-10-04 by EulalieCoevoet