SofaDefrost/SoftRobots #214: Can't compile Sofa

ClaraDoms, 2023-05-04:
Hi @EulalieCoevoet
I'm an internship at CRIStAL and i'm currently working on CoBra and I have some issues while compiling Sofa with SoftRobots. In fact, my simulation not only need SoftRobots but also SoftRobotsInverse and I have 2 problems with this. Firstly, after the compilation with SoftRobots, it is said that the plugin SoftRobots is not founded (https://imgur.com/a/7QUR9lP) and secondly, the SoftRobotsInverse I have is not up to date. Can you help me please ?

alxbilger, 2023-05-22:
Hi,
Can you confirm that you did compiled SoftRobots in addition to SOFA?
If you did, make sure that the binary of the plugin is somewhere on your computer. Ideally, it must be in the same folder than the SOFA binaries.

alxbilger, 2023-12-13:
Closing due to inactivity

closed 2023-12-13 by alxbilger