SofaDefrost/SoftRobots #223: Extra createTopologyHandler  call in tutorials

dusevitch, 2023-06-14:
You guys have made some absolutely fantastic tutorials. They've been super helpful!

I'm seeing this error when I run the Tripod example for `step4-3.py`:

```
[ERROR]   [UniformMass(mass)] TopologyData: indices already has a TopologyDataHandler. createTopologyHandler should only be called once at init of the TopologyData.
[WARNING] [UniformMass(mass)] totalMass value overriding the value of the attribute vertexMass. 
vertexMass = totalMass / nb_dofs. 
To remove this warning you need to set either totalMass or vertexMass data field, but not both.
```

It seems like it's coming when calling `self.ElasticBody.init()` in  to the Elastic function, as it calls the UniformMass function in elasticbody.py and seems to be conflicting with some other definition of the vertexMass in another function. I'm  trying to figure out where the vertexMass or other UniformMass is also defined in the code here or behind the scenes where I'm not seeing it?

dusevitch, 2023-06-15:
Awesome, thanks @alxbilger!

EulalieCoevoet, 2023-10-20:
I see that https://github.com/sofa-framework/sofa/pull/3882 has been merged. Should we close this issue ?