SofaDefrost/SoftRobots #225: Having problems loading the Tutorial of Tripod

Cuberkk, 2023-06-26:
Hi guys,

Recently I am trying to load the tutorial of Tripod in SOFA, while every files will have a problem like this:

[ERROR]   [SofaPython3::SceneLoader] Unable to completely load the scene from file 'C:/Users/comid/SoftRobots-master/examples/tutorials/Tripod/myproject/step1.py'.
Python exception:
  ModuleNotFoundError: No module named 'stlib3'

At:
  C:\Users\comid\SoftRobots-master\examples\tutorials\Tripod\myproject\step1.py(8): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(848): exec_module
  <string>(6): <module>

I have added the plugins of SoftRobot, SoftRobot_Inverse and STLIB. And I have added their path to the system variable SOFA_EXTERNAL_DIRECTORIES.

I think the reason why I can't compile the scene might be I missed some steps in loading the plugins. Do you have any ideas?

Cuberkk, 2023-06-26:
Never mind. I fixed it by adding stlib3 site-packages to python_path :)

closed 2023-06-26 by Cuberkk

datou1209, 2023-07-07:
How did you do that?