SofaDefrost/SoftRobots #226: Having trouble opening step 4 of Tripod Tutorial

Cuberkk, 2023-06-28:
Hi guys,

When I am trying to open Step 4-1 of Tripod Tutorial, there will be a warning in the terminal:

[WARNING] [SceneCheckMissingRequiredPlugin] This scene is using component defined in plugins but is not importing the required plugins.
  Your scene may not work on a sofa environment with different pre-loaded plugins.
  To fix your scene and remove this warning you just need to cut & paste the following lines at the beginning of your scene (if it is a .scn):
  <RequiredPlugin name="Sofa.Component.AnimationLoop"/> 
  <RequiredPlugin name="Sofa.Component.Constraint.Lagrangian.Correction"/> 
  <RequiredPlugin name="Sofa.Component.Constraint.Lagrangian.Solver"/> 
  <RequiredPlugin name="Sofa.Component.IO.Mesh"/> 
  <RequiredPlugin name="Sofa.Component.LinearSolver.Direct"/> 
  <RequiredPlugin name="Sofa.Component.Visual"/> 
  <RequiredPlugin name="Sofa.GL.Component.Rendering3D"/> 
  <RequiredPlugin name="Sofa.GUI.Component"/> 

and showing nothing in SOFA.
After I added these plugins into the plugin-list, the warning disappeared while there was still nothing shown in SOFA.
(The same thing happens to Step 4-2 & Step 4-3 files)
Anyone have any ideas about how to fix this?

alxbilger, 2023-06-29:
Hi,

could you host your files directly in this conversation? They are currently not shared on gdrive, and if you delete them, people won't be able to understand your question.

Also, tell us how you use SOFA (binaries or compiled from sources?), and the versions of SOFA and SoftRobots. Thanks.

Cuberkk, 2023-07-03:
Hi,

Sorry for the delay responding and confusion.

I am using a SOFA binaries (V22.12.00) on the Windows. And I just use the SoftRobots plugins in the plugin_files of the V22.12.00 SOFA binaries.

So when I run the Step 4-1, what appears in my SOFA interface is:
[image]
and what appears in the terminal is:
[image]
Do you have any ideas what might go wrong?

alxbilger, 2023-07-04:
The log in the terminal does not bring any information on the actual problem. Do you have any other error message before what you shared?

Can you check in the `View` tab in the interface that something is checked (Force Fields for example)?

Where did you get the scene file. Is this a scene shared with the SOFA binaries or did you get them from GitHub?

alxbilger, 2023-07-04:
The scenes in the `myproject` folder are for you to write (your are in a tutorial). If you want the final scene, go in the `details` folder.

Cuberkk, 2023-07-05:
Got it!
I got it work on my computer. There are some warning on the terminal while the scene can be animated.
[image]

alxbilger, 2023-07-06:
If you get the latest version of the scene on GitHub, the error should go away.

Cuberkk, 2023-07-08:
Got it. Thanks!

closed 2023-07-08 by Cuberkk