SofaDefrost/SoftRobots #229: Sofa.PythonScriptController

Emanuele-n, 2023-07-17:
Hello,
I am trying to run the example pneunetgripper-tuto.py , by copying the final SOFA scene simulation of the soft pneumatic Gripper from  the Appendix at pneunetgripper-tuto.md. However, when running with "runSofa -l path/to//libSofaPython3.so ./pneunetgripper-tuto.py"  I get the following error:
```
[ERROR]   [SofaPython3::SceneLoader] Unable to completely load the scene from file './pneunetgripper-tuto.py'.  
Python exception:   
  ValueError: Object type PythonScriptController<> was not created  
The object 'PythonScriptController' is not in the factory.  
But the following exits:  
                      : SutureController (63% match)  
                      : DataController (61% match)  

```
Can you please help me?

alxbilger, 2023-07-17:
Hi,

unfortunately the text of this tutorial is based on an old version of the SofaPython plugin. The component `PythonScriptController` no longer exists in SofaPython3. I suggest to have a look on `examples/tutorials/PneunetGripper/details/step7-grabTheCube.py`, which is an updated version of the final scene. In this scene, the controller is in another file (`examples/tutorials/PneunetGripper/details/wholeGripperController.py`) and is a Python class derived from `Sofa.Core.Controller`. This way, the new class can be used as a SOFA component: 

```python
rootNode.addObject(WholeGripperController(node=rootNode))
```

Emanuele-n, 2023-07-18:
Thanks a lot!

closed 2023-07-18 by Emanuele-n