SofaDefrost/SoftRobots #240: SurfacePressureConstraint leading to unsuitability in simulation

Macavitycode, 2023-09-08:
Hey everyone,

Background:

I'm new to SOFA and soft robotics, so in case this issue is a result of my naivete I apologize. 

I followed through the tutorial for a pneunet gripper from this repository [link](https://github.com/SofaDefrost/SoftRobots/tree/master/examples/tutorials/PneunetGripper) and got it working. Using the provided script as reference, I made a new model of an LSOVA, similar to one shown in documentation [here](https://project.inria.fr/softrobot/documentation/constraint/surface-pressure-constraint/).  

Mesh generation: 

Body: I've designed the meshes using FreeCAD and GMSH as shown in [this](https://project.inria.fr/softrobot/documentation/from-design-to-mesh-generation-using-freecad-and-gmsh/) tutorial. 

[image]

Cavity: After generating the body mesh in FreeCAD, I split it into components using the mesh split tool, which gave me two components: the outer mesh and the inner cavity. I exported only the cavity as an STL file.

[image]

I put this into GMSH and got a  .msh file.

Issue:

I've tried to simulate the script from the tutorials, modifying the sources of meshes, removing the stiff layer that's specific to the pneunet, and modifying the boundary condition. Apart from these, no changes regarding structure of the code is made.

When SurfacePressureConstraint is initialized for the cavity, it loads as long as the value parameter is set to 0. 

[image]
 
When I try to increase or decrease the pressure the mesh immediately implodes.

[image]

I am unsure of why this is happening and any help/guidance would be greatly appreciated. 

My meshes and code is linked [here](https://github.com/Macavitycode/docker-sofa/tree/master/code) , specifically sim2.py and LSOVAs directory.

Thank you so much for the help

EulalieCoevoet, 2023-09-08:
Hello, 

Try this: 

```python
cavity.addObject(
        "SurfacePressureConstraint",
        name="SurfacePressureConstraint",
        template="Vec3",
        valueType="pressure",
        value=0.0025,   # This should actually be value = pressure * dt 
        triangles="@cavityMesh.triangles",
        flipNormal=True  # Invert the faces' orientation of your mesh, your cavity is inside out 
    )
```

Macavitycode, 2023-09-08:
Hey, thank you, that seemed to fix a lot of instabilities. I still get the same issues when I change the pressure by too much, but I believe thats an issue with my design or simulation's other parameters. Thank you for your help again.

closed 2023-09-08 by Macavitycode