SofaDefrost/SoftRobots #268: errors for building softrobots plugin

Parabolakit, 2024-03-01:
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/constraint/model/PositionModel.h:30,
                 from ../src/SoftRobots/component/constraint/model/PositionModel.inl:33,
                 from ../src/SoftRobots/component/constraint/model/PositionModel.cpp:29:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/constraint/UnilateralPlaneConstraint.h:36,
                 from ../src/SoftRobots/component/constraint/UnilateralPlaneConstraint.inl:33,
                 from ../src/SoftRobots/component/constraint/UnilateralPlaneConstraint.cpp:29:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/constraint/model/CableModel.h:31,
                 from ../src/SoftRobots/component/constraint/CableConstraint.h:33,
                 from ../src/SoftRobots/component/constraint/CableConstraint.inl:30,
                 from ../src/SoftRobots/component/constraint/CableConstraint.cpp:30:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/constraint/model/CableModel.h:31,
                 from ../src/SoftRobots/component/constraint/model/CableModel.inl:33,
                 from ../src/SoftRobots/component/constraint/model/CableModel.cpp:30:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/constraint/model/SurfacePressureModel.h:33,
                 from ../src/SoftRobots/component/constraint/SurfacePressureConstraint.h:33,
                 from ../src/SoftRobots/component/constraint/SurfacePressureConstraint.inl:30,
                 from ../src/SoftRobots/component/constraint/SurfacePressureConstraint.cpp:29:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/constraint/model/SurfacePressureModel.h:33,
                 from ../src/SoftRobots/component/constraint/model/SurfacePressureModel.inl:34,
                 from ../src/SoftRobots/component/constraint/model/SurfacePressureModel.cpp:31:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.h:32,
                 from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.inl:33,
                 from ../src/SoftRobots/component/behavior/SoftRobotsConstraint.cpp:30:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.cpp:30:
../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.inl: In instantiation of ‘void softrobots::forcefield::PREquivalentStiffnessForceField<DataTypes>::init() [with DataTypes = sofa::defaulttype::StdRigidTypes<3, double>]’:
../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.cpp:39:36:   required from here
../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.inl:119:17: warning: ignoring return value of ‘constexpr bool sofa::type::Mat<L, C, Real>::invert(const sofa::type::Mat<L, C, Real>&) [with unsigned int NbLine = 6; unsigned int NbColumn = 6; <template-parameter-2-3> = void; unsigned int L = 6; unsigned int C = 6; real = double]’, declared with attribute nodiscard [-Wunused-result]
  119 |                 m_CInv[n].invert(m_complianceMat[n]);
      |                 ^~~~~~
In file included from ../framework/DefaultType/src/sofa/defaulttype/typeinfo/TypeInfo_Mat.h:25,
                 from ../framework/DefaultType/src/sofa/defaulttype/DataTypeInfo.h:35,
                 from ../framework/Core/src/sofa/core/objectmodel/BaseData.h:25,
                 from ../framework/Core/src/sofa/core/objectmodel/Data.h:25,
                 from ../framework/Core/src/sofa/core/objectmodel/Base.h:25,
                 from ../framework/Core/src/sofa/core/objectmodel/BaseContext.h:25,
                 from ../framework/Core/src/sofa/core/objectmodel/BaseObject.h:24,
                 from ../framework/Core/src/sofa/core/behavior/StateAccessor.h:26,
                 from ../framework/Core/src/sofa/core/behavior/BaseForceField.h:26,
                 from ../framework/Core/src/sofa/core/behavior/ForceField.h:25,
                 from ../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.h:31,
                 from ../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.inl:30,
                 from ../src/SoftRobots/component/forcefield/PREquivalentStiffnessForceField.cpp:30:
../framework/Type/src/sofa/type/Mat.h:695:34: note: declared here
[... 38 lines cut ...]
                 from ../src/SoftRobots/component/constraint/model/AffineFunctionModel.h:33,
                 from ../src/SoftRobots/component/constraint/model/AffineFunctionModel.inl:30,
                 from ../src/SoftRobots/component/constraint/model/AffineFunctionModel.cpp:32:
../src/SoftRobots/component/behavior/SoftRobotsBaseConstraint.h:125:45: error: ‘virtual sofa::type::vector<std::__cxx11::basic_string<char> > softrobots::behavior::SoftRobotsBaseConstraint::getBaseConstraintIdentifiers()’ marked ‘override’, but does not override
  125 |     virtual sofa::type::vector<std::string> getBaseConstraintIdentifiers() override final
[... cut ...]
make[2]: *** [external_directories/SoftRobots/CMakeFiles/SoftRobots.dir/build.make:128: external_directories/SoftRobots/CMakeFiles/SoftRobots.dir/src/SoftRobots/component/constraint/PositionConstraint.cpp.o] Error 1
make[2]: *** [external_directories/SoftRobots/CMakeFiles/SoftRobots.dir/build.make:180: external_directories/SoftRobots/CMakeFiles/SoftRobots.dir/src/SoftRobots/component/constraint/model/CableModel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:14145: external_directories/SoftRobots/CMakeFiles/SoftRobots.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

Parabolakit, 2024-03-01:
I have no ideas how to modify the source file to successfully build the plugin.

alxbilger, 2024-03-01:
Someone also had the error here: https://github.com/sofa-framework/sofa/discussions/4014#discussioncomment-6404552

You need to have matching versions of SOFA and SoftRobots. See the first warning paragraph in the documentation: https://www.sofa-framework.org/community/doc/plugins/build-a-plugin-from-sources/

Parabolakit, 2024-03-01:
> Someone also had the error here: [sofa-framework/sofa#4014 (reply in thread)](https://github.com/sofa-framework/sofa/discussions/4014#discussioncomment-6404552)
> 
> You need to have matching versions of SOFA and SoftRobots. See the first warning paragraph in the documentation: https://www.sofa-framework.org/community/doc/plugins/build-a-plugin-from-sources/

Thanks!!! It works

closed 2024-03-01 by alxbilger