SofaDefrost/SoftRobots #304: Issue on Cablegripper Tutorials Step 0

yu-mei, 2024-09-18:
Dear,

I am expolring to run the step0.py in Tutorials/CableGripper/details/step0 to see the overall simulation. But I cannot see the finger/gripper model and got the following issue related to Mesh:

**Environment**
Ubuntu 22.04 + SOFA_24.06.00 binary 

**Results:**
[image]

**Issues**
```
[INFO]    [SceneCheckerVisitor] Finished validating node "root".
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field rotation: np.float64(0.0) np.float64(0.0) np.float64(105.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field translation: np.float64(20.0) np.float64(0.0) np.float64(0.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field rotation: np.float64(0.0) np.float64(0.0) np.float64(105.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field translation: np.float64(20.0) np.float64(0.0) np.float64(0.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field rotation: np.float64(180.0) np.float64(0.0) np.float64(65.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field translation: np.float64(-10.0) np.float64(0.0) np.float64(-4.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field rotation: np.float64(180.0) np.float64(0.0) np.float64(65.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field translation: np.float64(-10.0) np.float64(0.0) np.float64(-4.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field rotation: np.float64(180.0) np.float64(0.0) np.float64(65.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [MeshVTKLoader(loader)] Could not read value for data field translation: np.float64(-10.0) np.float64(0.0) np.float64(34.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field rotation: np.float64(180.0) np.float64(0.0) np.float64(65.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field translation: np.float64(-10.0) np.float64(0.0) np.float64(34.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field rotation: np.float64(0.0) np.float64(0.0) np.float64(0.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field scale3d: np.float64(1.0) np.float64(1.0) np.float64(1.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field translation: np.float64(0.0) np.float64(0.0) np.float64(0.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field rotation: np.float64(0.0) np.float64(0.0) np.float64(0.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field scale3d: np.float64(20.0) np.float64(20.0) np.float64(20.0) 
[WARNING] [OglModel(OglModel)] Could not read value for data field translation: np.float64(0.0) np.float64(0.0) np.float64(0.0) 
[INFO]    [SceneCheckerVisitor] Validating node "root" with checks: [SceneCheckCollisionResponse, SceneCheckDeprecatedComponents, SceneCheckDuplicatedName, SceneCheckMissingRequiredPlugin, SceneCheckUsingAlias, SceneCheckMultithreading]
[WARNING] [SceneCheckMissingRequiredPlugin] This scene is using component defined in plugins but is not importing the required plugins.  
  Your scene may not work on a sofa environment with different pre-loaded plugins.  
  To fix your scene and remove this warning you just need to cut & paste the following lines at the beginning of your scene:   
  root.addObject('RequiredPlugin', name='Sofa.Component.AnimationLoop') # Needed to use components [FreeMotionAnimationLoop]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Collision.Detection.Algorithm') # Needed to use components [BVHNarrowPhase,BruteForceBroadPhase,CollisionPipeline]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Collision.Detection.Intersection') # Needed to use components [LocalMinDistance]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Collision.Geometry') # Needed to use components [LineCollisionModel,PointCollisionModel,TriangleCollisionModel]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Collision.Response.Contact') # Needed to use components [RuleBasedContactManager]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Constraint.Lagrangian.Correction') # Needed to use components [LinearSolverConstraintCorrection,UncoupledConstraintCorrection]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Constraint.Lagrangian.Solver') # Needed to use components [GenericConstraintSolver]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Engine.Select') # Needed to use components [BoxROI]  
  root.addObject('RequiredPlugin', name='Sofa.Component.LinearSolver.Direct') # Needed to use components [SparseLDLSolver]  
  root.addObject('RequiredPlugin', name='Sofa.Component.LinearSolver.Iterative') # Needed to use components [CGLinearSolver]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Mapping.Linear') # Needed to use components [BarycentricMapping]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Mapping.NonLinear') # Needed to use components [RigidMapping]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Mass') # Needed to use components [UniformMass]  
  root.addObject('RequiredPlugin', name='Sofa.Component.ODESolver.Backward') # Needed to use components [EulerImplicitSolver]  
  root.addObject('RequiredPlugin', name='Sofa.Component.SolidMechanics.FEM.Elastic') # Needed to use components [TetrahedronFEMForceField]  
  root.addObject('RequiredPlugin', name='Sofa.Component.SolidMechanics.Spring') # Needed to use components [RestShapeSpringsForceField]  
  root.addObject('RequiredPlugin', name='Sofa.Component.StateContainer') # Needed to use components [MechanicalObject]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Topology.Container.Constant') # Needed to use components [MeshTopology]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Topology.Container.Dynamic') # Needed to use components [TetrahedronSetTopologyContainer]  
  root.addObject('RequiredPlugin', name='Sofa.Component.Visual') # Needed to use components [VisualStyle]  

[SUGGESTION] [SceneCheckMultithreading] This scene is using components implemented sequentially while a parallel implementation is available. Using the parallel implementation may improve the performances. Here is the list of sequential components in your scene and their parallel equivalent:   
        BVHNarrowPhase -> ParallelBVHNarrowPhase  
        BruteForceBroadPhase -> ParallelBruteForceBroadPhase  
        TetrahedronFEMForceField[Vec3d] -> ParallelTetrahedronFEMForceField[Vec3d]  

[INFO]    [SceneCheckerVisitor] Finished validating node "root".
```

I have tried to implement the plugins into the code but still not working. I am wondering if anything wrong with _MeshVTKLoader(loader)_.

Best,
Mason

yu-mei, 2024-09-18:
This issue can be solved by downgrade the numpy to 1.26.4 version following Discussion: #298. But I am wondering if there are other ways to solve it since numpy have already been at least 2.0.0.

mirzazz10, 2024-09-26:
@yu-mei , were you able to control the gripper?
 Thanks for the this discussion by the way, I ran into the same issue.

yu-mei, 2024-09-27:
> @yu-mei , were you able to control the gripper? Thanks for the this discussion by the way, I ran into the same issue.

@mirzazz10 Yes I can use the keyboard ctrl+arrow to move the gripper. However, I feel it cannot realize open/close the grippers following the guidance. After checking the grippercontroller.py, I have not seen the corresponding code. Besides that, everything works perfectly.