SofaDefrost/SoftRobots #307: Issues on Tutorials/CableGripper/step5

yu-mei, 2024-10-03:
Hi folks,

I found I can successfully run step 4 to use keyboard to control the single finger, but I cannot control the grippers (three fingers). There is the issue bug shown in the terminal:

```
[ERROR]   [FingerController(FingerController)] Unable to execute code.  
Python exception:  
AttributeError: Unable to find attribute: cable
``` 

Anyone have any experiences or ideas on this issue?

Thanks,

yu-mei, 2024-10-03:
Just found in the finger.py, adding `self.cable = args[0]` will fix this issue.

closed 2024-10-03 by yu-mei

EulalieCoevoet, 2024-10-04:
Thanks for reporting your problem. I've had a look at it and it seems like this line 14 `self.cable = args[0]` has always been [there](https://github.com/SofaDefrost/SoftRobots/blob/38e0f8d5d400f9c85221fd291bd84ce7af57f238/examples/tutorials/CableGripper/details/finger.py#L14). Maybe you removed it by mistake? I'm glad it works for you anyway.

yu-mei, 2024-10-04:
> Thanks for reporting your problem. I've had a look at it and it seems like this line 14 `self.cable = args[0]` has always been [...]

It might be my mistake. Thank you for checking, Eulalie.