SofaDefrost/SoftRobots #337: Seemingly missing functions in softrobots.actuators

AdelGuiot, 2025-08-04:
Hello,

I am trying to pick up a soft robotics simulation started by someone else. Here is the link to the project: https://github.com/pchaillo/ucl_stiff_flop

The file that defines the properties of the soft robot ("Stiff_Flop_CLASS.py" in the project) calls constrain from softrobot.actuator. Constrain seems to contain a function called ConstrainCavity(points, parents, indices, tolerance, name). However, from what I can see in the latest version of the softrobot library, "constrain" is not a subpart of softrobot.actuator, and I cannot find the ConstrainCavity function anywhere.

Do you happen to know where it might have been moved? Thank you in advance!

alxbilger, 2025-08-04:
I cannot find the file `Stiff_Flop_CLASS.py` in the mentioned repository. So I cannot really help you.
But I can tell you that actuator is a concept from the plugin `SoftRobots.Inverse` (https://github.com/SofaDefrost/SoftRobots.Inverse), not `SoftRobots`.

alxbilger, 2025-08-04:
The missing module is in @pchaillo's own fork: https://github.com/pchaillo/SoftRobots/tree/master/python3/softrobots/actuators

AdelGuiot, 2025-08-05:
Hello, thank you very much for finding that out, I failed to consider that he might have forked the repository!

The script is still not operational (it's erroring because "Softrobots.Inverse is missing" even though it is used in my scene, which is weird since the plugin comes preinstalled with SOFA, so I'm looking into that), but the issue regarding softrobots.actuators is now fixed. Thanks for your help!